Printrbot/Printrhub #4: add git tags for firmware release numbers.

bmv437, 2016-12-10:
Hi there. I can see that my printrhub has firmware package 108, but I'm not sure at what commit that firmware package was built. Is it possible to add git tags so we can keep track of what code gets shipped in which firmware build?

```json
{
    "firmware": 108,
    "fw_cloud_buildnr": 108,
    "fw_info_sent": true,
    "fw_update": true,
    "ipaddress": "192.168.1.10",
    "locked": false,
    "mk20": true,
    "name": "printrbot",
    "serialnr": "",
    "softapip": "0.0.0.0",
    "ssid": "myWifi"
}
```

Thanks!

bmv437, 2016-12-10:
So it looks like the cloud firmware version is 108, but in this repo it's version 106 (at least that's what the header files say)

https://github.com/Printrbot/Printrhub/blob/master/esp/src/config.h#L35
https://github.com/Printrbot/Printrhub/blob/master/mk20/src/framework/core/Application.h#L65

Is this the latest code running on the PrinterHub? or did `FIRMWARE_BUILDNR` just not get updated?

mickbalaban, 2016-12-10:
Hey Brandon, the reason for this version not matching the one that is deployed on printers is that we were transitioning from private repo with multiple branches, into this one. Phillip was working on preparing the repo for release, and in the meantime I pushed a new firmware update that had few bug fixes. 

I will merge the code asap and make sure everything compiles correctly and then it will be up to date. Feature that is missing in this repo is addition of Light control. Previously we had "update" button in control panel, which was removed and replaced with light.

mickbalaban, 2016-12-12:
ok, code should be now in sync with old repo. All future updates will be done in this repo going forward...

mickbalaban, 2016-12-12:
repo is now tagged. Please reopen if you see any other issues.

closed 2016-12-12 by mickbalaban