Printrbot/Printrhub #7: Load filament workflow - include unload?

ururk, 2016-12-13:
When I go to change filament, it would be nice if the load filament workflow could include an unload step:

* Heat extruder
* Question prompt: "Is filament already loaded??"
  * If `Yes`, proceed to unload filament
* Once done, resume standard load process

Right now this is a two-step process, which includes two extruder heat-up times.

mickbalaban, 2016-12-13:
Thanks for your suggestion, but I disagree that your what you proposed is better user experience. Currently you are prompted to either load or unload filament. If you unload it first, and then click on load filament, hotend is not cooled off completely, so second time it will take less time to heat up. Still very simple.. one click, one action.

What you are proposing is that instead of selecting "load" or "unload", we do one "change filament" action, heat the hotend, then ask user if they want to load or unload. Now we went from one click to two clicks for something we needed only one before.

You are also assuming that user wants to change filament, which may be wrong. What if they just want to remove the roll of filament and store it in container after printing? With your approach, they would unload the filament, then printer will go into load filament after that, so they will have to cancel out of it. Again, adding clicks and making it more complicated.

Sorry, but I am not convinced that this is better then existing way of doing it.

ururk, 2016-12-13:
True. Though I wasn't suggesting removing the unload filament step - I suppose this suggestion would require a third option (Change Filament), which could be confusing since the other steps would still be present. I guess I want it both ways - load/unload as distinct steps, and a change option. Feel free to close this suggestion out.

mickbalaban, 2016-12-13:
no worries. We ran scenarios like this when we were designing the interface and tried to minimize user clicks with each action. Our choices may not have been the best so please feel free to continue to submit suggestions, and even better, start playing with the code :)
Ill keep this open to see if anybody else has any opinion on this.

ururk, 2016-12-13:
Thanks - good to know. I really like the simplicity of the setup, and it prevents people from getting confused when they go to do something.

As for playing around with the code - wish I had the time!

closed 2017-02-14 by mickbalaban