Printrbot/Printrhub #19: Attempting to print model produces "Unknown Error" or "File Not Found"

pr3sidentspence, 2017-03-14:
1. Import project from thingiverse
2. Send to printer
3. Select part and tap print
4. Shows downloading briefly
5. Get error on screen. Either "Unknown Error" or "File Not Found"

I can't say that I've seen this from manually uploaded STLs, but I might have. I haven't done this much yet.

wjsteele, 2017-10-19:
This appears to be a slicing failure.  Go to the printrbot.cloud page and look to see if the filename is in red... if it is, then the slicer failed for some reason... most likely it is outside the limits of the print area.  Scale or move the part so that it is within the printable envelope and try again.  (I deleted the old jobs from the printer and re-loaded them after I fix the object.)

pr3sidentspence, 2017-10-20:
Hmm. I don't think that was the case for me.