Printrbot/Printrhub #25: Z maxes out at end of print causing jam

sfcgeorge, 2017-10-14:
I'm using the small starter spool you ship with the printer. When a print finishes the gantry whizzes up the Z at warp speed, crashes into the spool, jams and causes a horrible grinding noise from the stepper as it skips! 

The spool naturally gets pulled to the left past where it's supposed to be because those mini spools have a weird lip in the middle but not at the sides so there's nothing to stop it from sliding over too far.

Don't have the gantry go so high, or ship the mini spools with printed inserts that add a lip to the outside so it doesn't slide over, or make the flange on the spool handle bigger.

sfcgeorge, 2017-10-20:
Maybe it isn't the Printrhub doing this but the end Gcode. It only seems to happen with the default calibration cuboid print, the others don't move up at the end.

ururk, 2017-12-19:
I designed a little add-on to prevent this from happening:

https://www.thingiverse.com/thing:1954463