Printrbot/Printrhub #35: Document G-Codes used in current firmware

pschuster, 2018-01-24:
G-Codes are hardcoded in the firmware (see Printr.cpp). This has issues with sending files via WiFi and for overall flexibility. 

As a first step we should document current G-Codes used and then decide where to split them or where to implement them. Documentation is required in [Wiki](https://github.com/Printrbot/Printrhub/wiki/G-Codes)

There are various possibilities:

1) Keep them in the firmware as is (hard coded)
2) G-Codes downloaded from the Cloud contain all G-Codes (i.e. G-Codes are generated in the cloud)
3) G-Codes are stored on internal SD card.
4) ...

## Tasks
- [ ] Document G-Codes used in current firmware
- [ ] Discuss and decide where to put them (in this issue)
- [ ] Create issues required for refactoring

oddirmeyer, 2018-01-25:
Starting G-code descriptions updated and ready for review.

pschuster, 2018-01-25:
Awesome. Thank you very much. That makes things easier I think. The next step would be to cluster those G-Codes in basic setup, and print preparation or should we do things like reset x and y and probing before each print?

abdrumm, 2018-04-04:
I am ok w minimum viable product that works ;) so we don’t HAVE to do it before every print. I think since all printers probe before print, it’s just best practice... conservative but not necessary